NARUTOfzr/Neptune_3_Pro_Plus_Max #3: Wiring diagram for Neptune 3 Pro control board

kozross, 2022-12-24:
Could we please have a wiring diagram for the ZNP Robin Nano_DW V2.2?

NARUTOfzr, 2022-12-24:
[image]
See if it's what you need.

[NEPTUNE 3 PRO User Guide.pdf](https://github.com/NARUTOfzr/Neptune_3_Pro/files/10298202/NEPTUNE.3.PRO.User.Guide.pdf)

kozross, 2022-12-24:
Would it be possible to have a labelled schematic with the pinouts? I am aware of this image, but it doesn't give me all the information I need.

NARUTOfzr, 2022-12-26:
There may not be the kind of pin image you need, but hopefully this will help.
[ZNP Robin Nano_DW V2.2 SCH.pdf](https://github.com/NARUTOfzr/Neptune_3_Pro/files/10302526/ZNP.Robin.Nano_DW.V2.2.SCH.pdf)

kozross, 2022-12-26:
@NARUTOfzr - thanks, that helps. Would it be possible to have a pinout diagram as well? Something like [this image](https://i.imgur.com/duvCECL.jpeg).

NARUTOfzr, 2022-12-26:
> @NARUTOfzr - thanks, that helps. Would it be possible to have a pinout diagram as well? Something like [this image](https://i.imgur.com/duvCECL.jpeg).
Yes, there is no such diagram at the moment. Sorry.

kozross, 2022-12-29:
Does the `X_EN` pin mean 'X endstop' or 'X enable'?

Peaceman1995, 2023-04-23:
Is there any parts list available? The schematics help, but I need to know what type of transistor/mosfet is used for the parts cooling fan? The one on my board is broken, and I'd like to repair it.
[image]

NARUTOfzr, 2023-04-24:
@Peaceman1995 You can find the information you need here.
https://github.com/elegooofficial/neptune3pro/tree/main/PCBA/Mainboard
[image]
[image]

Peaceman1995, 2023-04-24:
Awesome, exactly what I was looking for. Many thanks :)