NARUTOfzr/Neptune_3_Pro_Plus_Max #7: Screen Unresponsive _ Cannot Print - Screen Will Not Update Firmware

cmeltoo, 2023-01-03:
Hello, I have used my neptune 3 pro for 2 weeks now without problems. It was printing great, then the screen froze in the middle of my last print and now the screen either "LCD RECOVERY" to that screen when I unplug the screen and plug it back in OR it loads to the home screen but does not respond to my touch, is useless. I tried upgrading firmware but I received the error "the system detects multiple tft files, cannot be upgraded, please remove extra tft file" and then starts up to the main menu with no responsiveness. I assure you I only have the 1 tft file and I formatted the card several times in between tries. 

My printer is bricked, how can I fix this and prevent it from ever happening again? :(

NARUTOfzr, 2023-01-03:
You can also try this on a Mac:
1. Format TF card again.
2. Move the file to the root of the TF card.
[elegooui_20221122.zip](https://github.com/NARUTOfzr/Neptune_3_Pro/files/10334286/elegooui_20221122.zip)

4. Open it using the Archive Utility that comes with your Mac. (Unpack the screen firmware file into the root directory of the TF card)
5. Directly pull out the TF card and insert the screen card slot to restart and upgrade.

NARUTOfzr, 2023-01-03:
You can try to format TF card using windows system and follow the instructions to update firmware. Because this is the Mac system caused by duplicate files. It's not visible on a Mac.

cmeltoo, 2023-01-03:
Thank you, this worked. I was able to install the screen firmware after putting the tft file on using a windows computer at work. What the hell, Mac? I assume this is why I sometimes see double the gcode files when printing from the sd card I use on my Mac. 

Do you know what causes the printer to brick? How do I avoid this from happening again?

NARUTOfzr, 2023-01-04:
Yes. The Mac system automatically creates some hidden files that you can't view on a MacBook. But other systems can.
I don't have a good way to solve this problem. I know from the feedback that not all Macbooks will have this problem.

martinstoehr, 2023-01-12:
Pretty easy fix on Mac: open Terminal-Window and navigate to the root-dir of the TF card.
Then type "rm ._*"
Now all hidden AppleDouble-files are removed and its working perfect.