NARUTOfzr/Neptune_3_Pro_Plus_Max #8: Display upgrades

afruzu, 2023-01-03:
Good morning, I'm trying the Neptune 3 PRO display to print and I would like to point out two changes to the display that I consider indispensable:
1) in the initial screen and in those for adjusting the printer parameters it would be essential to have, in addition to the indication of the temperatures of the bed and the nozzle, also the current coordinates, at least the indication of the z and the indication flashing WHEN NECESSARY the home procedure as Marlin do on other screen display
2) in the print screen it is necessary to invert the PAUSE and STOP PRINTING keys because the LED button is too close to this second key and printing can be stopped by mistake

afruzu, 2023-01-03:
I'm sorry, there is another relevant indication to be highlighted in the display screen and this is when the nozzle and bed heating resistors are on and when they are not and not only looking for temperature value changing

afruzu, 2023-01-03:
These information are visible in every textual 12864 Marlin display and is incredible that are not visible in this lcd grafic display

NARUTOfzr, 2023-01-04:
@afruzu Thanks for your feedback. Relevant improvement suggestions have been fed back to the R&D department.
I don't think there will be a wrong stop in question 2, because it needs to be confirmed before stopping.

Thanks again!

afruzu, 2023-01-06:
I apologize if I add the observations on the necessary additions to the display a little at a time, but these derive from the practice of using the printer that I received only a few days ago.
since I massively use layers with variable height, I find it necessary that in the screen relating to the printing phase in progress the value of the z coordinate must be indicated with two decimal digits, i.e. in hundredths of a millimeter and not only with a single decimal digit, i.e. in tenths of a millimetre. this would make the quota reached in the printing phase much more readable.

MULAWI, 2023-07-07:
It not difficult, the marlin screen is more useful even not that pretty like the elegoo one.

- Z Offset should be accessible without starting the leveling menu
- XYZ Steps should be visible and changeable 
- Linear Advance setting visible and changeable 

thats all what i’m missing ☺️