NARUTOfzr/Neptune_3_Pro_Plus_Max #10: will there be support Linear advance?

chukey, 2023-01-13:
Neptune 3 have Linear advance, when support in PRO version firmware?

TheFeralEngineer, 2023-01-16:
It's in there, but the boards are still running 2225 drivers, which have the same shortcomings as the 2208 in that they aren't so great with linear advance. I think there have been some strides made in getting it to work, but none of the test patterns i've tried to produce have even extruded a single line of filament.

sergiitk, 2023-01-20:
According to the [chaneglog](https://github.com/NARUTOfzr/Neptune_3_Pro/blob/ff579325a7b81ccf9ed37305520ad58b61d63fc3/Update%20Log-Neptune3Pro%26Plus%26Max.docx) it was added in 1.1.5.1 board firmware. I was successful in tuning it up, even for several different filaments. There's definitely a huge difference in the print quality. No more corner blob when printing a calibration cube.

TheFeralEngineer, 2023-01-20:
> According to the [chaneglog](https://github.com/NARUTOfzr/Neptune_3_Pro/blob/ff579325a7b81ccf9ed37305520ad58b61d63fc3/Update%20Log-Neptune3Pro%26Plus%26Max.docx) it was added in 1.1.5.1 board [...]

I was able to get it to produce a part, but none of the test patterns worked when I tried. I switched to Klipper shortly after.

SleepinDevil, 2023-01-22:
> I was able to get it to produce a part, but none of the test patterns worked when I tried. I switched to Klipper shortly after.

How did you transition to Klipper on the stock board that comes with the N3PRO? Did you flash it with the firmware that is floating around unlocking Klipper on the N3 non-pro?

Has using Klipper on the stock board and this printer been smooth sailing? I'm assuming the touchscreen becomes moot with Klipper then, I am assuming it is not even displaying anything on the touchscreen anymore?

I need linear advance working pretty badly because the under/over extrusion from change in print speed is really grinding me down in my quest for perfection!

TheFeralEngineer, 2023-01-22:
> How did you transition to Klipper on the stock board that comes with the N3PRO? Did you flash it with the firmware that is floating around unlocking Klipper on the N3 non-pro?
>
> Has using Klipper on [...]

Check my github repo. I have everything you need to run Klipper as an almost plug and play for the pro. It's the same chip as the standard with the exception of one pin (servo pin on the old board is the signal pin for the inductive sensor). The bin file is the same, the cfg file was 95% similar and I've got a handful of people running them reliably

valkuc, 2023-03-02:
Hello. I'm using Neptune 3 Plus, flashed last firmware 1.2.5.1a and I'm don't see any differences on calibration cube. Tryed K factor from 0.0 to 2.0. Am I doing something wrong?

ChrisCopter, 2023-04-26:
> According to the [chaneglog](https://github.com/NARUTOfzr/Neptune_3_Pro/blob/ff579325a7b81ccf9ed37305520ad58b61d63fc3/Update%20Log-Neptune3Pro%26Plus%26Max.docx) it was added in 1.1.5.1 board [...]

What are your settings mate?

MULAWI, 2023-07-07:
> Hello. @NARUTOfzr I'm using Neptune 3 Plus, flashed last firmware 1.2.5.1a and I'm don't see any differences on calibration cube. Tryed K factor from 0.0 to 0.2. Am I doing something wrong?
> 
> UPD: Looks like the differences become visible after 0.2 and above.

I did print the pattern on Ellis page but all looks same… surprisingly the 0 looks best 🙀 have a Neptun 3 Plus

valkuc, 2023-07-11:
I'm running the latest firmware and can confirm that LA works, but with the out-of-the-box head the result is not very efficient. I updated the head as described in the description for this video - https://www.youtube.com/watch?v=lpzA1QHWDTI - and the LA factor became noticeable. For example, right now I'm using K0.084 for PETG plastic.

aanaumkin, 2023-08-06:
> Я использую последнюю прошивку и могу подтвердить, что LA работает, но с головкой из коробки результат не очень эффективен. Обновил голову как описано в описании к этому видео - https://www.youtube.com/watch?v=lpzA1QHWDTI - и стал заметен фактор LA. Например, прямо сейчас я использую K0.084 для пластика PETG.

Dear colleague, can you please tell me what is the latest version of 1.1.5.1a? If on 1.1.5.1a there is no possibility to customize PID Bed in it, so I switched to 1.1.5.3 in it this G-Code construction is available. I printed Linear Advance v1.2K calibration model from Dmitry Sorkin and did not see LA on it. Either it is the firmware or it is really necessary to Upgrade the print head. Thanks in advance for the answer.