NARUTOfzr/Neptune_3_Pro_Plus_Max #13: Problem with

machmalpause, 2023-01-19:
I have just updated the board to V 1.1.5.3 which went fine.
I then tried to update the screen to V 1.5beta. The update constantly fails at 65% with a red write error. Rebooting the screen without the SD card fails with a data error. Worst, I can't revert back to V1.4, it also fails at 65%.
Please help !

NARUTOfzr, 2023-01-20:
Hi @machmalpause .Usually updates firmware without breaking the screen board. I repackage a firmware. You need to reformat the TF card or replace another TF card.

[3D36.zip](https://github.com/NARUTOfzr/Neptune_3_Pro/files/10462951/3D36.zip)

machmalpause, 2023-01-20:
Hi,

Thank you very much for your ultra fast response and help, much appreciated !

Unfortunately, your zipped firmware file "3D36.tft" did not fix the problem. Flashing the firmware fails consistently at 65% with a blue screen and a red "write error" message, the screen reboots with a blank screen and a red "Data ERROR!".

I have already tried this repeatedly with several different freshly formatted (FAT) microSD cards of different size (Sandisk/Kingston 2, 8, 64 GB), no luck at all. Flashing the previous V1.4 firmware file also fails with identical error. However, previously (i.e. last week) updating from V 1.2 to V 1.4 was no problem at all. Odd.

Any other ideas how to get the screen functional again ?

With your latest (and future ?) firmware improvements I hope the Neptune 3 Pro will become the gold standard for budget FDM printing very soon  ;-)

Thanks and best wishes from Germany !

machmalpause, 2023-02-01:
Apparently the PCB of the screen was faulty. Elegoo kindly offered a free replacement which works perfectly fine. Thank you !

closed 2023-02-01 by machmalpause