NARUTOfzr/Neptune_3_Pro_Plus_Max #15: Filament Change start to print layer, then stops

s-eckey, 2023-01-19:
When using the filament change, the printer does not stop right away but starts to print the next layer and stops then.

steps to reproduce:
1. slice a stl with filament change
2. print

[image]

red arrows: the printer began to print the layer where the filament change should happen
green arrows: after filament change

(blob problem already reported: https://github.com/NARUTOfzr/Neptune_3_Pro/issues/14 )

NARUTOfzr, 2023-01-20:
#12 #14 
@s-eckey Yes, the filament does not stop immediately when it changes or pauses. Because the master needs to process the cached code.
Thank you for your feedback! It is not normal to extrude an extra filament when resuming after suspension. We will solve this problem after the Spring Festival. Thank you again.

ditchy1, 2023-01-27:
1.1.5.2 handled a filament change in Cura but 1.1.5.3 extracts the filament before starting to resume the print.

NARUTOfzr, 2023-02-02:
@s-eckey @ditchy1 
I found this error in the test, and I am very sorry for the trouble brought to your use.

This firmware fixes this error. Increased use of gcode-M355 to control led. The Z automatic compensation function is optimized.
[1.1.5.3a.zip](https://github.com/NARUTOfzr/Neptune_3_Pro/files/10567702/1.1.5.3a.zip)

krayon, 2023-12-17:
This issue still occurs in 1.3.5.1c - the printer still _starts_ printing the next layer *BEFORE* pausing and requesting filament change.