NARUTOfzr/Neptune_3_Pro_Plus_Max #16: Beta 1.5 display firmware - permanent wifi NODE module

afruzu, 2023-01-20:
I want to thank the staff for having considered the need to update (with the 1.5 beta) the display firmware in particular by showing the coordinates of the nozzle in the initial screen and for having increased the number of decimal showed of the z coordinate so that the height of the print layer is more precisely perceivable. 
Still with regard to the display, I would like to mention that this informations should be visible not only in the general screen and in that of the printing phase, but also in that relating to "preparation" which now are not visible! 

I would also like to remind the technicians that, instead of rather useless and redundant animations concerning the large icons of the various pages, a simple flashing light or a little animation of the thermal icons would be essential to understand when the nozzle and the plate are in the active heating phase - when resistors are on or when temporarily off, for normal thermal hysteresis loop. This information helps a lot to understand what the printer is doing at any moment.

In order of the use of the port for the micro SD card, I evidence that it is not always very easy to use it! whose position it's although frontal and it's difficult continually put and remove the micro SD card from the printer slot. So I thought of adding permanently an external wifi module to the Neptune 3 PRO printer (in particular the excellent Fiberpunk's NODE for Ender 3 model).

With this system permanently plugged into the SD slot and USB port, I have some problems recognizing the presence of the SD card from the printer display. For that, I have also started a reporting post to Fiberpunk's technical assistance and I am waiting for the matter to be resolved but I also report it to the ELEGOO technicians so that they can consider the problem.

When using the NODE wifi module in a rather permanent installation of the device, when I try to print some of the files loaded on the micro SD card in the rear of NODE, not through the procedure on the web page 192.168.1.x:88 but directly from printer touch screen menu, I get a “card not present” error :-(

Just to keep the NODE in a permanent installation on the printer, the visibility of the contents of the rear SD card should be guaranteed both from the web page and also from the print menu of the printer itself. 

I represented this problem so that it can be kept in mind during the revision of the firmware of both the printer and the NODE.

afruzu, 2023-02-03:
Thanks, I noticed that with the new firmware 1.1.5.3a the problem I had reported about the visibility of the files contained in the SD card located behind the Fiberpunk NODE has been resolved!
I remind those who need it that to see the contents of the SD card you must first unmount the NODE.
I also remember that, unlike what Fiberpunk support communicated to me, the protocol that works on my Elegoo Neptune 3 PRO printer is the SDIO method and not the SPI method.
This issue can be closed.

NARUTOfzr, 2023-02-03:
Hi @afruzu , thank you so much for your detailed feedback. As for the use of NODE WiFi module, we have not tried to use this module at present, so we cannot accurately support such problems, but we will continue to test and optimize the firmware.

As for the screen control, I think you have a point. Currently, the screen firmware is still being further optimized. Please look forward to the new version of UI.