NARUTOfzr/Neptune_3_Pro_Plus_Max #17: Problem den mit dem Upgrade der TFT Firmware

Manuel2171, 2023-01-20:
Hallo,

Ich habe ein Problem mit dem TFT Update. Hinweis, es wurden mehrere TFT-Versionen gefunden, update nicht möglich. Danach startet der TFT Bildschirm und ich habe nur noch  Zugriff auf den Button Links oben, alle anderen lassen sich nicht mehr ansteuern

Manuel2171, 2023-01-20:
Baud: 9600
SD Card Update...
The system detects mulitple tft files, can not be upgraded, please romove the extra file

Manuel2171, 2023-01-20:
Problem there was just one tft file on the sd card.

machmalpause, 2023-01-20:
Hi, das ist mir auch passiert, nachdem ich das firmware file mit macOS die microSD Karte geschrieben hatte. MacOS legt beim Kopieren auch eine versteckte Datei an, ein sog. "hidden file" beginnend mit ".". Diese Datei musst du löschen, dann funktioniert's !

NARUTOfzr, 2023-01-20:
@Manuel2171 @machmalpause I've made a video on this subject that I hope will help you.
https://youtu.be/mdb4PTPlJh4

machmalpause, 2023-01-20:
Hi, thanks.
I'm well aware of this video (and it was very helpful for my first screen update) but Manuel2171 apparently wasn't.

Manuel2171, 2023-01-20:
Thank you, I will try tomorrowManuel Strauß manuel-strauss.de015233904909

Manuel2171, 2023-01-21:
Problem solved, großer Dank @machmalpause . Die versteckten Files waren es, da hätte ich ja auch mal selber draufkommen können. Danke Dir :)

Manuel2171, 2023-01-21:
> 

Thank you, Problem solved

Manuel2171, 2023-01-21:
> Hi, das ist mir auch passiert, nachdem ich das firmware file mit macOS die microSD Karte geschrieben hatte. MacOS legt beim Kopieren auch eine versteckte Datei an, ein sog. "hidden file" beginnend mit ".". Diese Datei musst du löschen, dann funktioniert's !

Danke für die Hilfe funkt jetzt alles wieder