NARUTOfzr/Neptune_3_Pro_Plus_Max #19: After filament got changed, the filament is getting retracted while the head is moving back to continue the print

OrgovanGeza, 2023-02-01:
I use the filament runout sensor, and it works great.
After I change the filament and I press Continue, it extrudes a little extra filament to make sure that everything is OK. 
However, while it moves back to the the position where it paused, the filament is getting continously retracted. 
Based on my observation it retracts exactly the same length as it moves on the X axis on it's way back.

Because of this when the print continues there is no extrusion in the next ~5 seconds.

Firmware: 1.1.5.3
UI Version: 1.4

https://user-images.githubusercontent.com/84339427/216177345-7b7318c7-149f-44d6-8308-770259f83887.mp4

NARUTOfzr, 2023-02-02:
@OrgovanGeza 
Thank you for your detailed feedback. I found this error in my test. I'm sorry for the trouble.

This firmware fixes this error. Increased use of gcode-M355 to control leds. The Z automatic compensation function is optimized.
[1.1.5.3a.zip](https://github.com/NARUTOfzr/Neptune_3_Pro/files/10567627/1.1.5.3a.zip)

OrgovanGeza, 2023-02-03:
I installed the firmware, and the issue is gone.
Thank you for the quick help!

closed 2023-02-03 by OrgovanGeza