NARUTOfzr/Neptune_3_Pro_Plus_Max #20: 1.1.5.3a firmware boot error

afruzu, 2023-02-03:
After the latest firmware update to version 1.5.3.a there is an anomalous behavior of the lighting LEDs (that light up at half power) during the boot phase when the printer is switched on! 
They must be turned off manually when finished. The software selector in the settings menu appears positioned on off, so when you turn it on, the LEDs go to full power and only then can they be turned off by returning the selector to off.
Finally, as regards the display, I would like to point out once again that the indication of the z coordinate to the third decimal digit, i.e. to the thousandth of a millimeter (0.000 mm) is completely wrong: the presence of the last digit, as well as being completely unreliable for the general tolerances of the machine, is often the cause of confusion.
It may seem like a fussy detail, but if the indication stops at the hundredth of a millimeter (0.00 mm) it is much more legible and reliable in consideration of the precision of the permitted movements.

afruzu, 2023-02-03:
在最新的固件更新到版本 1.5.3.a 之后，当打印机开启时，在引导阶段 LED 灯（以半功率点亮）出现异常行为！
完成后必须手动关闭它们。 设置菜单中的软件选择器显示在关闭位置，因此当您打开它时，LED 会全功率运行，只有这样才能通过将选择器返回到关闭来关闭它们。
最后，关于显示，我想再次指出，z 坐标到小数点后第三位，即千分之一毫米 (0.000 mm) 的指示是完全错误的：最后一位数字的存在， 以及对于机器的一般公差完全不可靠，通常是造成混淆的原因。
这似乎是一个繁琐的细节，但如果指示停在百分之一毫米（0.00 毫米）处，考虑到允许的运动精度，它会更加清晰可靠。

NARUTOfzr, 2023-02-03:
@afruzu   
Thank you for your feedback.     
 .5.3a firmware enables M355 gcode to control LED. Turn on the light when starting the machine, you can effectively judge the motherboard has started normally. In future improvements, we will try to turn the lights off after a few seconds of blinking to indicate that the motherboard has started.

afruzu, 2023-02-03:
very good, excellent thing this M355 gcode.
However, if it is possible to adjust the intensity of the LEDs via software, it is not enough to have only an on/off button on the display!