NARUTOfzr/Neptune_3_Pro_Plus_Max #21: PID Bed tuning

fr-oz-t, 2023-02-05:
Hiya,

I have recently updated to the lastest firmware and screen gui, however i still cannot PID tune the heated bed on the elegoo neptune 3 pro? 

PID tuning the hotend works a treat.
PID tuning bed gives PID Autotune failed! Bad heater id

NARUTOfzr, 2023-02-21:
很抱歉。官方的固件不支持热床PID。因为开启热床pid会导致电压不稳定，LED灯会有闪烁的现象。如果您不在意LED闪烁，你也许可以尝试自行编译固件：https://github.com/NARUTOfzr/Elegoo-Neptune-marlin2.1.1

byFops, 2023-02-23:
could you compile me a current firmware with hot bed PID tuning? I do not use the LED lighting.

Is it difficult to compile an own firmware version with hot bed PID tuning and can your firmware be used as a template?

Thanks for your help

byFops, 2023-02-23:
> @byFops <https://github.com/byFops>
> Of course. You're welcome. I will compile the firmware for you when I
> arrive at the company tomorrow morning.
>
> Compiling your own firmware is not difficult. There are currently many
> tutorials on YouTube on how to build marlin firmware using 'VScode', which
> is a good process to learn.

Perfect. Then I will work in and try to compile my own firmware. Thanks for
your quick help

byFops, 2023-02-23:
n homepage I only get the one for the Neptune 2.

Thanks for your help

NARUTOfzr, 2023-02-24:
@byFops 

[1.1.5.1x-PID-BED-20230224.zip](https://github.com/NARUTOfzr/Neptune_3_Pro_Plus_Max/files/10820186/1.1.5.1x-PID-BED-20230224.zip)
This firmware enables the hotbed PID.

byFops, 2023-02-24:
Hello Naruto, thanks for the firmware. Unfortunately this is for the Neptune 3 Pro. I have the Neptune 3 Plus. So the settings for the print bed do not fit. Could you compile the firmware again for the 3 Plus. Preferably on the basis of the firmware 1.2.5.3a.

Thanks already

BirdPriest, 2023-03-01:
Hey @NARUTOfzr sorry for jumping in on here, would it be possible for me to grab that firmware enables the hotbed PID.(Neptune3Pro)? Thank you for your time.

NARUTOfzr, 2023-03-01:
> Hey @NARUTOfzr sorry for jumping in on here, would it be possible for me to grab that firmware enables the hotbed PID.(Neptune3Pro)? Thank you for your time.
@BirdPriest Of course.
(Neptune3Pro)
[1.1.5.3x-PID-BED-20230301.zip](https://github.com/NARUTOfzr/Neptune_3_Pro_Plus_Max/files/10862791/1.1.5.3x-PID-BED-20230301.zip)

BirdPriest, 2023-03-01:
@NARUTOfzr Thank you so much!

NARUTOfzr, 2023-03-13:
[1.2.5.1x-PID-BED-20230313.zip](https://github.com/NARUTOfzr/Neptune_3_Pro_Plus_Max/files/10958864/1.2.5.1x-PID-BED-20230313.zip)
The difference between 1.x.5.1a and 1.x.5.3a is that Z displays different values. Subsequent motherboard firmware may be compatible with different screen firmware.

byFops, 2023-03-13:
Thanks for the information and quick help :-)

hobytomy, 2023-04-17:
Hello NARUTOfzr,
Hello Naruto, could you please provide me again from the current version Neptune3 MAX a version with "Pid-BED-On"?

Thanks for your help
Greetings from Germany please keep up the good work with your devices and services! you are really good !
Thomas

ChrisCopter, 2023-04-26:
> Hey @NARUTOfzr sorry for jumping in on here, would it be possible for me to grab that firmware enables the hotbed PID.(Neptune3Pro)? Thank you for your time.
> @BirdPriest Of course.
> (Neptune3Pro) [...]

Is this the latest full release version for the Pro, only with additional Bed-Tuning, or are there any other changes made to it?

gbarbehenn, 2023-10-20:
OK, so PID isn't used by default.  What is used?
The Neptune 3 config.h has:
/**
 * PID Bed Heating
 *
 * If this option is enabled set PID constants below.
 * If this option is disabled, bang-bang will be used and BED_LIMIT_SWITCHING will enable hysteresis.
 *
 * The PID frequency will be the same as the extruder PWM.
 * If PID_dT is the default, and correct for the hardware/configuration, that means 7.689Hz,
 * which is fine for driving a square wave into a resistive load and does not significantly
 * impact FET heating. This also works fine on a Fotek SSR-10DA Solid State Relay into a 250W
 * heater. If your configuration is significantly different than this and you don't understand
 * the issues involved, don't use bed PID until someone else verifies that your hardware works.
 * @section bed temp
 */
//#define PIDTEMPBED

//#define BED_LIMIT_SWITCHING

both commented out?