NARUTOfzr/Neptune_3_Pro_Plus_Max #22: WIfi and Input Shaping

Mordevolt, 2023-02-07:
Hi! I have 2 questions:
1. Any chance to see official support of MKS wifi module, since it is slot for it on the motherboard? Talking about this device: https://aliexpress.ru/item/32816882694.html?gatewayAdapt=glo2rus&sku_id=64718626331
2. Are you going to implement Marlin 2.1.2 and input shaping officially? It is already implemented by community: https://github.com/dderg/Marlin-Neptune-3-Pro

NARUTOfzr, 2023-02-21:
@Mordevolt 
1. I think there will be tutorials on this in the future, but it should be clear that WiFi is not included in the default configuration of the machine. So it's not within the scope of technical support.
2. Reintegrating a new firmware may take more time to test, but I will report back to the developers. Thank you for sharing.

NARUTOfzr, 2023-02-21:
> @Mordevolt
>
> 1. I think there will be tutorials on this in the future, but it should be clear that WiFi is not included in the default configuration of the machine. So it's not within the scope of [...]

dmorehead1988, 2023-05-30:
Hi @NARUTOfzr ,

Is there any update with the tutorial for adding wifi module? Looking at forums, there are quite a few users that would like this functionality.

ama564, 2023-10-12:
Any luck on input shaping?

ama564, 2023-10-24:
Any update in input shaping

ama564, 2024-02-25:
Guess no updates.  Seems like elegoo stopped caring about np3