NARUTOfzr/Neptune_3_Pro_Plus_Max #23: Offset keeps changing

boehser-enkel, 2023-02-10:
Hi,
the offset keeps changing. Also between prints directly after another with the same file.
Is this a software bug?

greetings

boehser-enkel, 2023-02-13:
Not the case when i run ABL before each print. Shouldn't it get stored for the next print?

NARUTOfzr, 2023-02-21:
Hi @boehser-enkel. The zoffset set during printing can be applied in the next printing, but will not be saved to EEPROM and will be restored to the Zoffset value set during tuning after restarting the machine. You can add M500 to the end of the gcode file and save the EEPROM after printing. You can also click the "Level" button after printing, and click the "Back" button in the upper left corner after entering the level interface to save EEPROM.