NARUTOfzr/Neptune_3_Pro_Plus_Max #24: Rolling back the firmware

br0cker, 2023-02-20:
Hi. 

I have updated the screen firmware to 1_5beta_20230114.tft. But in this firmware, the fans on the head do not turn on during printing. I wanted to bring back the firmware, but when updating I get a message: Model does not match, TJC4827X243_011C, File info ERROR! 
I tried [V1_4_1_20230210.tft]  [elegooui_20221122.tft]
And I still get this message. Please help me.

Neptune 3 Pro

BenKmann, 2023-02-20:
Hi, 
my Fan starts working at around the third layer. Have you tried waiting?

NARUTOfzr, 2023-02-21:
@br0cker 
1. Download the firmware package and decompress it to obtain the firmware.
[image]

2. Format TF card to FAT32 4096 bytes
3. Update the screen firmware

NARUTOfzr, 2023-02-21:
> Hi, my Fan starts working at around the third layer. Have you tried waiting?

This is usually set by the slicing software:
[image]

br0cker, 2023-02-21:
> 

Hi. 
Thank you very much. I did it. 
My mistake was that I did not download the entire archive, but the files separately. 
Good luck.

br0cker, 2023-02-21:
> Hi,
> my Fan starts working at around the third layer. Have you tried waiting?

Hi. 
Thanks for the reply. I was scared, I thought it was because of the new firmware. It turned out that everything is exactly as you write. Have a nice day.

closed 2023-02-21 by br0cker