NARUTOfzr/Neptune_3_Pro_Plus_Max #25: Neptune 3 Pro bootlooped on "Update firmware" screen

Bipopski, 2023-03-01:
Today when I turned on my printer, after "booting" screen, it showed "Update firmware" screen (I did not upload any file to update firmware to microSD card, at this stage only files on card were .gcode files). I waited about 10 minutes, nothing changed. I turned off printer, formatted microSD card and uploaded "V1_4_1_20230210.tft" and "ZNP_ROBIN_NANO.bin" (1.1.5.1a) files to it. Firstly I tried to update firmware on screen, as in manual and I got error that this is wrong version. After that, I tried to update firmware on board, it stared as it should and again it was bootlooped on "Update firmware" screen for more then 10 minutes. I turned off power and I checked files on microSD card. there was file "ZNP_ROBIN_NANO.CUR", which (according to manual) means that update went successfully, but it is still bootlooped on "Update firmware" screen. What should I do to be able to use this printer again? Please help.
[image]

NARUTOfzr, 2023-03-01:
Now you can try to remove the TF card and restart the machine.

Bipopski, 2023-03-01:
I tried the same process with "1_5beta_20230114.tft" and "ZNP_ROBIN_NANO.bin" (1.1.5.3), and now it worked out as it suppose to. I don't know what I did wrong before.

NARUTOfzr, 2023-03-01:
I have received feedback that this can happen when file names have special symbols. We have also repeated the problem. Software engineers are still working on the fix. For now, please try to exclude the gcode file that is causing the error.

Bipopski, 2023-03-01:
Machine is working ok now.

closed 2023-03-01 by Bipopski

NARUTOfzr, 2023-03-01:
> Machine is working ok now.

Thank you for your feedback!