NARUTOfzr/Neptune_3_Pro_Plus_Max #26: 'Resume printing

anpotdar, 2023-03-01:
I have multiple gcode files on the SD card. I accidentally unplugged the printer while it was printing a design further down in the

NARUTOfzr, 2023-03-01:
@anpotdar I'm sorry to hear that. Could you please provide the gcode file of TF card and the gcode file being printed? Also provide the firmware version number, we will try to reproduce the problem, and then try to fix it. Thank you for your time.

anpotdar, 2023-03-04:
@NARUTOfzr attached gcode. _filamentbox_top_raft_ is the print that was interrupted. The printer resumed printing _chopper_p2-1_white_ after I confirmed resume on the screen. At first glance it looks like the printer automatically picked the first alphabetically ordered file on the SD card.

Firmware version 1.1.5.3a; UI version V1.4.1.

[Neptune_3_Pro_Plus_Maxissues26.zip](https://github.com/NARUTOfzr/Neptune_3_Pro_Plus_Max/files/10889888/Neptune_3_Pro_Plus_Maxissues26.zip)