NARUTOfzr/Neptune_3_Pro_Plus_Max #30: N3Pro not displaying thumbnails

anpotdar, 2023-03-08:
Firmware version 1.1.5.3a
UI version V1.4.1
Slicer Ultimaker Cura v5.2.2
N3Pro profile imported from https://github.com/NARUTOfzr/Cura-Neptune-TEST
Plugin installed - MKS Wifi Plugin 1.4.2
Created post-processing script for 'Create Thumbnail' 128x128

The thumbnail is viewable in windows explorer using the PowerToys file explorer extensions i.e. the slicer is generating the thumbnail and being read through the gcode but the printer does not parse it

jaegerschnitzel, 2023-03-09:
I think it's only supported with Elegoo Cura.
I also opened a bug report for the MKS Wifi Plugin but without response: https://github.com/Jeredian/mks-wifi-plugin/issues/365

jaegerschnitzel, 2023-05-29:
This is working for me :)
https://github.com/finict/mks-wifi-plugin-N3

Please verify and close the issue.

anpotdar, 2023-06-03:
@jaegerschnitzel, I followed the steps outlined in the above link and I get the following error when I try to save the rendered gcode file:
```
2023-06-02 23:38:50,571 - DEBUG - [MainThread] MKSWifiPlugin.MKSPreview.add_preview [111]: mks_simage value: 160
2023-06-02 23:38:50,579 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]: Exception: Unable to write to file EN3PRO_socal accents logo v13: [Errno 2] No such file or directory: 'D:\\Desktop\\N3\\bin\\new_size.txt'
2023-06-02 23:38:50,585 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]: Traceback (most recent call last):
2023-06-02 23:38:50,585 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "UM\Qt\Bindings\OutputDeviceManagerProxy.py", line 150, in _writeToDevice
2023-06-02 23:38:50,586 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     device.requestWrite(nodes, file_name, limit_mimetypes, file_handler, **kwargs)
2023-06-02 23:38:50,586 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "D:\Software\UltiMaker Cura\UltiMaker Cura 5.3.1\share\uranium\plugins\LocalFileOutputDevice\LocalFileOutputDevice.py", line 139, in requestWrite
2023-06-02 23:38:50,586 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     self._performWrite(file_name, selected_type, file_handler, nodes)
2023-06-02 23:38:50,587 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "D:\Software\UltiMaker Cura\UltiMaker Cura 5.3.1\share\uranium\plugins\LocalFileOutputDevice\LocalFileOutputDevice.py", line 160, in _performWrite
2023-06-02 23:38:50,587 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     self.writeStarted.emit(self)
2023-06-02 23:38:50,587 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "UM\Signal.py", line 216, in emit
2023-06-02 23:38:50,587 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     self.__handleEmitIndirect(*args, **kwargs)
2023-06-02 23:38:50,588 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "UM\Signal.py", line 323, in __handleEmitIndirect
2023-06-02 23:38:50,588 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     self.__performEmit(*args, **kwargs)
2023-06-02 23:38:50,588 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "UM\Signal.py", line 361, in __performEmit
2023-06-02 23:38:50,588 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     signal.emit(*args, **kwargs)
2023-06-02 23:38:50,588 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "UM\Signal.py", line 216, in emit
2023-06-02 23:38:50,590 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     self.__handleEmitIndirect(*args, **kwargs)
2023-06-02 23:38:50,590 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "UM\Signal.py", line 323, in __handleEmitIndirect
2023-06-02 23:38:50,590 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     self.__performEmit(*args, **kwargs)
2023-06-02 23:38:50,590 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "UM\Signal.py", line 353, in __performEmit
2023-06-02 23:38:50,591 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     func(*args, **kwargs)
2023-06-02 23:38:50,591 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "C:\Users\anike\AppData\Roaming\cura\5.3\plugins\MKSWifiPlugin\MKSWifiPlugin\MKSPreview.py", line 112, in add_preview
2023-06-02 23:38:50,591 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     screenshot_string += add_screenshot_str(image, simage, simage, ";simage:")
2023-06-02 23:38:50,592 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "C:\Users\anike\AppData\Roaming\cura\5.3\plugins\MKSWifiPlugin\MKSWifiPlugin\MKSPreview.py", line 26, in add_screenshot_str
2023-06-02 23:38:50,592 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     with open(r'D:\Desktop\N3\bin\new_size.txt', 'w') as f:
2023-06-02 23:38:50,592 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]: FileNotFoundError: [Errno 2] No such file or directory: 'D:\\Desktop\\N3\\bin\\new_size.txt'
```

anpotdar, 2023-06-03:
Restored cura from a backup and the issue goes away. Redid the steps as listed in https://github.com/finict/mks-wifi-plugin-N3 and got the above error again