NARUTOfzr/Neptune_3_Pro_Plus_Max #32: Update in Progress blocked

giova609, 2023-03-12:
After trying to update the Neptune 3 PRO to firmware 1.1.5.1a it got stuck on "Update in Progress". Even if you remove the TF Card, "Update in Progress" still appears. Thank you

NARUTOfzr, 2023-03-12:
According to firmware update instructions. Try to download the entire zip package. Then format the TF card. Update the mainboard firmware.

giova609, 2023-03-13:
Thanks for the help, the zip file is asking me for a password.  I use Edge Chromiun with web browser.

NARUTOfzr, 2023-03-13:
Trying to reload the firmware 1.1.5.1a won't work either?

giova609, 2023-03-13:
I've tried both reloading 1.1.5.1a and the beta, but it gets stuck on "Update firmware..."

NARUTOfzr, 2023-03-14:
Hi@giova609 
Check whether the LED on the motherboard lights up? 
Maybe you can shoot a firmware update video to confirm the current situation.

giova609, 2023-03-14:
Thanks for the help. I solved. I reloaded all the firmware starting from 1.1.4.1 (present on the printer before the problem) up to 1.1.5.1a.

closed 2023-03-14 by giova609