NARUTOfzr/Neptune_3_Pro_Plus_Max #33: Here's the usual pain in the ass for the z digits in the display!

afruzu, 2023-03-13:
I loaded the yesterday card and display firmware: great job, however I noticed that on the printing phase screen, the z coordinate value in millimeters and that it has again returned to being shown only one decimal ! :-(

It was good to eliminate the other two coordinates x and y from visibility, as their continuously changing numerical value did not add any significant information, however z is fundamental and its precision helps to understand exactly the progress of the printing process.

I think I'm not the only one to use print profiles that include layers with a height of 0.16 mm (or 0.12 mm,  0.28 mm etc.). In all of these cases, as I have pointed out in the past, it would be very convenient to read the digit of z height approximated to one hundredth of a mm, with TWO decimal digits after the point and not with only ONE or THREE redundant, as Elegoo proposed. 
For heaven's sake, I know that what I'm asking is an absolutely marginal question, but I can't understand the reason for these choices!

NARUTOfzr, 2023-03-14:
Hi@afruzu Thank you very much for your continuous attention and feedback on the problems of machine use. I believe the new version of the firmware will optimize these issues.
This is the firmware file that is still being tested and I would be happy to share it with you first. Perhaps you can share some of your thoughts along the way.
We're going to continue to improve the UI.
[3Pro_1.x.5.1f.zip](https://github.com/NARUTOfzr/Neptune_3_Pro_Plus_Max/files/10970491/3Pro_1.x.5.1f.zip)

afruzu, 2023-03-14:
Tank you! my first impression on it is that you have made an excellent work!

afruzu, 2023-03-15:
I did some printing tests having updated the firmware with your beta versions, I found them excellent! Really perfect prints (Cura 5.3.0 slicer with the default profiles that Naruto has lately provided in the original package).

Even the display is now in place, complete clear and legible.   I would just like to point out that the display could still be perfected to highlight the thermal transients (for example flashing icons indicating the heating phase in progress for both the plate and the nozzle) and also for the geometric transients: in fact, when the printer  is in the phases prior to home (G28), a false z coordinate value is shown, while it would be preferable to show flashing symbols (such as ???) instead of an arbitrary inexact numerical value.