NARUTOfzr/Neptune_3_Pro_Plus_Max #34: Use screen with Octoprint

daunjo, 2023-03-16:
Hello,

Would it be possible to make the screen active when printing from Octoprint? Now its just idle even when printing.

Thanks,

/Jonas

daunjo, 2023-05-25:
@valkuc I do not get that behaviour on my printer with the latest firmware, can you explain what you did to make it work?

daunjo, 2023-05-25:
So now a working progress bar would be much appriciated. :)

valkuc, 2023-05-25:
> @valkuc Edit, working now when I did m75 in the terminal

You have to add custom GCODE commands in Octoprint. Something like this:

[image]

xqoperator, 2025-01-14:
Does M73 work for anyone on Neptune 3? (the issue is still open so it might not...)
If it does - on which firmware mainboard/lcd?
For me M75 does enable the progress screen but M73 does not update it.