NARUTOfzr/Neptune_3_Pro_Plus_Max #36: FW naming and latest version

lazarhypercube, 2023-03-18:
Can you let me know what is latest version of FW for 3 plus?

I see there is [1.2.5.1a-20230210] a week old, but in Historical version folder there is [1.2.5.3] that I am using right now and its 1 month old. Is 1.2.5.1a newer and better then 1.2.5.3?

Also for screen there is 1.4.1 that is update last week and also in Historical version 1.5 beta that is week old. 

In readme file you have:
1.x.4 - 1.x.5.1a historic version
1.x.5.1a Release version

Kinda confusing do have same version in historic and release :) I think you should work a bit on version system to be easier for everyone to keep track what version is latest one.

Keep up the great work.

LarsStegman, 2023-03-19:
Using the GitHub releases would be nice

Zartimus, 2024-03-15:
Also wondering about this historical folder. Lost in translation?