NARUTOfzr/Neptune_3_Pro_Plus_Max #39: 1.1.5.1a release firmware led wobblin problem

ramazandagli, 2023-03-21:
The problem is when i started print led brightnes wobbling. I checked cables and gcode all looks fine. When printer doesn't print anything leds are stable not brightnes wobbling at all. Is it normal or not ? Thanks for helps.

coddiewomple, 2023-03-21:
i that where my printer i would be getting a multimeter out to monitor the voltage into the led strip to see if there might be a power supply problem and maybe disconnect the led to see if it restores full printer function. It is possible for those smd leds to got funny after they get hot just like in a tv so sounds like you would be advised to contact elegoo support direct for that issue if it still happens with different firmware perhaps even stock fw.

ramazandagli, 2023-03-21:
I doubt the firmware because there was no problem in the stock version. Can you tell me if you have a recommended firmware version, thanks

closed 2023-03-23 by ramazandagli