NARUTOfzr/Neptune_3_Pro_Plus_Max #43: Filament runout with OctoPrint - head does not move to "home" for reload

autumnwalker, 2023-04-17:
With latest firmware OctoPrint will detect filament runout. I get a notification in Octo that it has detected filament runout with no option but to "X" the notification away.

I pulled out the old filament, replaced the spool, loaded new filament (all by hand, not using the Neptune screen) and then clicked "resume" in Octo - it seems to be working fine.

Expected Solution:
I expected when OctoPrint got a filament runout from the printer that it would have "parked" the head at the home location allowing me to unload / load the filament without keeping the hot end on the printed part.

I also expected that there would be some clear options in the dialog box notifiying me of filament runout to cancel, resume, etc. - maybe that's an Octo issue?

autumnwalker, 2023-04-25:
This is great - I will try that GCODE!

I am running the following:

Firmware: 1.1.5.1a
UI Version: 1.4.1

grahamr, 2023-04-26:
Got it!
now to test a bit more

it looks like (for me):

M412 returned
Recv: echo:Filament runout ON ; Host handling OFF

`M412 S1 H1` - resets the sensor and enables host handling

with H1 in Octoprint logs i see a pause, and the pause Gcode executes - nothing about a runout in the terminal, but it works as expected
`Changing monitoring state from "Pausing" to "Paused"`

with H0 in Octoprint i see:
`Recv: //action:prompt_end
Recv: //action:prompt_begin FilamentRunout T0
Recv: //action:prompt_show
Recv: //action:paused filament_runout 0
Printer signalled that it paused, switching state...
Changing monitoring state from "Printing" to "Pausing"
Recv: ok
Recv: ok
Changing monitoring state from "Pausing" to "Paused"`

but it does NOT execute the gcode to pause it

also i added this to my Resume GCode 

; reset runout
M412 S1

NARUTOfzr, 2023-04-26:
Hi @grahamr . Thank you for your test feedback. This should be a problem with the firmware source logic, we will try to reproduce the problem according to your description and try to optimize it. Thank you very much.

grahamr, 2023-05-24:
@NARUTOfzr i just upgraded from **1.1.5.1a** and **UI 1.4.1** to **1.1.5.1b** and **UI 1.4.2**
and now when i check my settings and compare with the `m503`, i noticed `m412` was not set, so i tried reading it, and get:

```
Send: M412
Recv: echo:Unknown command: "M412"
Recv: ok
```

MaliciousCodez05, 2023-10-08:
@NARUTOfzr Bumping this, I've had a ton of failed prints because the filament runout sensor doesn't work when printing with Octoprint. Every creality and prusa machine have worked just fine with this functionality, I would greatly appreciate having this one resolved. Thanks! 😅