NARUTOfzr/Neptune_3_Pro_Plus_Max #51: Where is firmware files?

valkuc, 2023-05-08:
Hi. I see you updated readme file with info regarding new firmware version (1.x.5.1b for board and 1.4.2 for screen), but I there is no downloadable files. Am I missing something?

See https://github.com/NARUTOfzr/Neptune_3_Pro_Plus_Max/tree/main/board%20firmware/Neptune3Plus

ChrisCopter, 2023-05-11:
Go to Code, than klick the green code button and download zip.

valkuc, 2023-05-11:
> Go to Code, than klick the green code button and download zip.

The last version in the repository is 1.2.5.1a-20230210, but in the readme file 1.x.5.1b is mentioned.

[image]

NARUTOfzr, 2023-05-11:
@valkuc @ChrisCopter Sorry for the confusion! This part of the file has not been uploaded, I will arrange as soon as possible.

ramdaron, 2023-05-19:
> @valkuc @ChrisCopter Sorry for the confusion! This part of the file has not been uploaded, I will arrange as soon as possible.

Please also upload the source code of the LCD. I plan to send a pull request with fixes in Russian localization.
Thank you!

NARUTOfzr, 2023-05-19:
@valkuc @ChrisCopter V1.4.2-1.x.5.1b ：Firmware has been uploaded.

@ramdaron The UI-HMI project file has been updated.     https://github.com/NARUTOfzr/Neptune_3_Pro_Plus_Max/tree/main/Source%20Code/LCD-TJC/V1.4.2

closed 2023-05-19 by valkuc