NARUTOfzr/Neptune_3_Pro_Plus_Max #53: Screen & Board Version Doesn't get updated

AhmedShaalan, 2023-05-11:
I have updated my machine to 1.4.1 for the screen, and 1.1.5.1a for the board. 
However, when i go to system info in the screen. it shows version 1.1.1 for the board and 1.1 for the screen.

NARUTOfzr, 2023-05-11:
@AhmedShaalan 
https://github.com/NARUTOfzr/Neptune_3_Pro_Plus_Max/blob/main/README-en.md
I think the screen firmware update failed, maybe refer to FAQ-8/9 to deal with it.
You can try updating the following firmware files.
[Neptune3pro-V1.4.2-1.1.5.1b.zip](https://github.com/NARUTOfzr/Neptune_3_Pro_Plus_Max/files/11411120/Neptune3pro-V1.4.2-1.1.5.1b.zip)

AhmedShaalan, 2023-05-11:
No, the update is successfully installed as per the message at the end of the update AND the screen originally didn't have animated Icons, now it has animated icons.

I will try to install the files you provided, but I can see that it is the beta version. Can i downgrade from the beta version later?

AhmedShaalan, 2023-05-16:
> @AhmedShaalan
> https://github.com/NARUTOfzr/Neptune_3_Pro_Plus_Max/blob/main/README-en.md
> I think the screen firmware update failed, maybe refer to FAQ-8/9 to deal with it.
> You can try updating the [...]

I have tried the files you have attached, first, I have inserted the TF card into the screen and it got updated successfully then the screen version showed 1.4.2 while keeping the Firmware at 1.1.1.

Then I inserted the TF card into the machine and it got updated. Now it's showing Firmware v1.1.5.1b, and UI v1.4.2

Thanks

closed 2023-05-16 by AhmedShaalan