NARUTOfzr/Neptune_3_Pro_Plus_Max #55: BIGTREETECH Module BTT TF Cloud V1.0 SD Cloud on neptune 3 max

eribertao, 2023-05-13:
I'm using this BIGTREETECH Module BTT TF Cloud V1.0 SD Cloud:
https://biqu.equipment/products/bigtreetech-module-btt-tf-cloud-v1-0-sd-cloud-wireless-transmission-module?variant=31555936944226
on my elegoo neptune 3 Max but when I upload files via wifi the printer does not read them. But if I turn the machine off and on again then they appear. There would be a way to be able to update the microsd card to avoid turning the printer off and on again. I tried to create a gcode file with m21 command but it doesn't work. Can you help me?

NARUTOfzr, 2023-05-19:
HI @eribertao  Try the latest firmware and see if it fixes the problem.

eribertao, 2023-05-20:
Do you say the latest firmware of the printer?

eribertao, 2023-05-21:
OK, thanks a lot! Now it works perfectly! I finally solved the problem of removing and inserting the microsd.