NARUTOfzr/Neptune_3_Pro_Plus_Max #58: Print Job Confirmation Language

AhmedShaalan, 2023-05-19:
Hi,

When I enable Multiple File Display option, then try to print I get the confirmation message in Chinese even if my language is set to English.
Also, I would like to ask, what is Multiple File Display😅

Thanks

[image]

Torrosua, 2023-05-22:
I can confirm from my end too.

Change language & reboot didn't help. It's always on Chinese

NARUTOfzr, 2023-05-24:
@AhmedShaalan @Torrosua An update has been made, attempting to reload UI firmware.

NARUTOfzr, 2023-05-24:
@AhmedShaalan It can support printing in folders, but I still recommend keeping printing files in TF cards to a minimum.

Prokofeb, 2023-06-03:
@NARUTOfzr  Is it possible to edit the translation in the display? In Russian, not everything is translated correctly.