NARUTOfzr/Neptune_3_Pro_Plus_Max #59: EEPROM datasize error

valkuc, 2023-05-24:
Sometimes after printing or after print cancel I'm getting "EEPROM datasize error" in Octoprint logs. The bed level is resetting and it's required to perform bed calibration again. It's quite annoying. You can find log file attached. What I did: I cancelled printing from Octoprint UI and then start raising up Z-axis using printer display UI. After that all got stuck.

[octoprint.log](https://github.com/NARUTOfzr/Neptune_3_Pro_Plus_Max/files/11554607/octoprint.log)

NARUTOfzr, 2023-05-24:
@valkuc  I have also reappeared the problem, which still needs some time to work out.

viruz82, 2024-03-05:
@NARUTOfzr some time means 10 months? Great support here. 👍🏼