NARUTOfzr/Neptune_3_Pro_Plus_Max #62: 1.x.5.1b多文件显示功能bug

chindev2012, 2023-05-27:
打印机型号：海王星3 Pro
主板固件版本：1.1.5.1b - 20230520
屏幕固件版本：1.4.2 - 20230520
通过屏幕高级设置-显示多个文件打开多文件功能
点击打印，打印结束或者中途停止打印后，再点击打印没有反应，无法继续打印

NARUTOfzr, 2023-05-28:
感谢您的反馈！为方便排除和修复该问题，能否提供一下卡里gcode文件让我们打印尝试？

chindev2012, 2023-05-29:
gcode是通过cura正常切片的文件，问题应该不会出在gocode文件上，可能我没有描述清楚，我举个例子，比如打印完成一个gocode后，准备打印下一个文件，这时点击屏幕上的打印按钮，只能听到点击按钮的声音，无法展示卡上的文件列表，只能通过重启机器解决

NARUTOfzr, 2023-05-30:
我复现了改问题:多文件显示开启后，在文件列表中，选中进入文件夹后直接操作返回，再次进入文件列表时根目录下大多数gcode文件不显示了（打印后的操作也是类似问题）。
我还发现通过再次点击文件夹，然后点左翻页可以返回到根目录下。
这个问题似乎是因为主板固件添加了进入文件列表时再次挂载SD卡导致。我们会排除此问题，再次感谢您的反馈，谢谢！

chindev2012, 2023-07-28:
请问，这个bug何时可以修复啊？这个功能挺好用的，另外是否还存在最多25个文件数量的限制？

NARUTOfzr, 2023-07-30:
嗨！多文件显示模式不限制文件数目。以下这个固件解决了多文件显式模式下文件缺失的问题。存在问题：多文件模式下，在文件夹内的文件无法预览缩略图或可能会显示错误的缩略图，根目录下的文件仍然可以正常显示缩略图，这并不影响打印，可以试试这个固件。

[Neptune_3_Pro_Plus_Max-firmware_1.x.5.1c-UI_V1.4.3.zip](https://github.com/NARUTOfzr/Neptune_3_Pro_Plus_Max/files/12208490/Neptune_3_Pro_Plus_Max-firmware_1.x.5.1c-UI_V1.4.3.zip)

chindev2012, 2023-07-31:
感谢您的分享，但是这版固件经过测试，进入一次打印目录以后再退出，就无法再次进入了，无论是否开启多文件显示选项，期待更好的更新！

NARUTOfzr, 2023-08-01:
您好！我尝试了并没有出现这种问题，请问屏幕固件和主板固件都更新了吗？能否帮忙拍段视频反馈看看，提前感谢！

chindev2012, 2023-08-01:
主板固件和屏幕固件都刷新了，我在tf卡里面考入了28个gcode文件，不知道是否和这个文件数量有关，第一次点击打印进入文件列表显示正常，返回主页面之后，再次点击打印就没有反应了，稍后我拍个视频

NARUTOfzr, 2023-08-01:
好的，感谢感谢！如果可以，文件不是太大的话，也麻烦将文件发给我试一试，我看看能不能复现，好反馈给软件工程人员优化。

chindev2012, 2023-08-01:
视频我已经发送到 qq 群里面了，但是我不清楚群里哪个是你

chindev2012, 2023-08-03:
昨天测试了一下，把文件名字改短，确实可以了👍