NARUTOfzr/Neptune_3_Pro_Plus_Max #66: How ready is the beta version?

christopheraryan, 2023-06-16:
I've been watching the repository and issues and was wondering how close the firmware was to a release status? When do you think it will be released?  And do you think it is stable enough for use, now?
Thanks!

jaegerschnitzel, 2023-10-20:
I think the development stopped because Neptune 4 is already available...

christopheraryan, 2023-10-21:
Seems that way. I'm a bit frustrated because it indicates the lack of commitment to their product line. I need to do more research on alternative companies and see if this is how they all are, or if it is just this company. The printer is fine, works well, etc but I'm starting to "outgrow" it. Now that I have more experience I want additional features (speed, better integration with Octopus, etc.). It seems that this company may not be the one to stay with.

ama564, 2023-10-24:
That's bullshit they need to make a reliable product first before releasing another one

gthb96, 2023-12-03:
I don't have a solution but have to say that Elegoo is one of the very few companies that even tries to support their products.
Atleast in that price category. Still a shame of course but as the firmwares work fine for me I'm not mad.