NARUTOfzr/Neptune_3_Pro_Plus_Max #68: Adding ESP8266 WI-FI module configuration

hindsa-nabeel, 2023-06-26:
Hi,
I understand that it is possible to plug in an ESP8266 WI-FI module (Makerbase MKS) to enable a WiFi connection, 
Is it possible to make available a WiFi configuration file or function within the firmware for that purpose?
Regards, Nabeel.

ShultsAlexey, 2023-07-04:
Мне тоже это нужно

miguelwerner, 2023-07-29:
I need this too.

hindsa-nabeel, 2023-07-30:
It has been over a month that this request was posted. I am not sure if NARUTOfzr reviewed it.
There is a new firmware posted 4 hours ago; with inconsistent version numbers in the changelog.
I would urge those who need or want this function to vote for this request.
Regards, Nabeel.

NARUTOfzr, 2023-07-30:
> 
Hello, everyone! @hindsa-nabeel   The new version of the firmware still cannot configure the 8266wifi module. I also hope to add these functions, but due to the rapid updating of the machine, most of our time and energy are focused on the new machine. We are sorry for not updating the firmware of the machine in time, and we are very sorry for not replying to your questions in time.
If you have any difficulties in using the machine, please contact our after-sales technical support in this email (3dp@elegoo.com), they will reply faster. Thank you for your support of elegoo products!

hindsa-nabeel, 2023-07-30:
Thank you @NARUTOfzr for your reply, 
We are very happy with Neptune 3 Plus quality and performance, but it would be great if you could add a configuration setting for "ESP8266" WI-FI module (if exists) in the menu.
It gives us confidence in Elegoo products when you indicate that you intend to include Wifi functionality in the firmware.
Kind regards, Nabeel.

hindsa-nabeel, 2023-09-26:
I was thinking about buying the new Elegoo (4) 3D printer; but I decided to hold on to ensure I am buying properly supported products before committing to newer ones. I am not sure I understand the difference between https://github.com/elegooofficial and https://github.com/NARUTOfzr. Who is responsible for regular maintenance and support?