NARUTOfzr/Neptune_3_Pro_Plus_Max #71: Printer resets itself when using the display to fast

Borotes, 2023-08-15:
The Neptune 3 Pro (latest firmware) resets itself when you are using it to fast..... for examble after printing you want to clean the nozzle, lift the extruder up 10cm and want to heat it up again, if you type too fast then, the os gets stuck and the printer reboots.

After that Z-Offest and everything else gets resetted.

staehle, 2023-08-18:
I had to downgrade back to the 1.4.1 and 1.x.5.1a my plus shipped with. The latest are way too buggy

ama564, 2023-11-05:
Same issue

coddiewomple, 2023-11-18:
yes my printer has this issue today after i created some folders on the SD card to organise the masses of model files on it currently trying to reverse that as unable to print at all when i press print button the machine just reboots and maybe 1 out of two times the eeprom is cleared also so have to go through levelling and z offset all over. Looks like this firmware is a bit rotten

coddiewomple, 2023-11-19:
I just levelled the bed again and upon exiting the levelling screen to return to main menu the machine reset itself also clearing eeprom. I give up where is support now the series 4 is out we have been abandoned with unfit for purpose proprietary firmware holding us hostage.