NARUTOfzr/Neptune_3_Pro_Plus_Max #74: Neptune 3 pro auto leveling problems.

PedroMarcos04, 2023-09-01:
I'm having problems with automatic leveling, the machine does the leveling, I adjust the z-offset but when printing it's uneven, in some places the nozzle scrapes the table, in others it's high and the filament doesn't adhere to the table.

FabioAmorim1501, 2023-09-07:
It sounds like the gantry is misaligned. Are your belts, screws and rollers tightened enough?

PedroMarcos04, 2023-09-07:
apparently everything is in order, the belt that synchronizes the two spindles of the Z axis is a little loose, the belt tooth may have dropped and one of the axes has lost synchronism with the other.  I even put a manual leveling kit on it, but haven't tested it yet.

BenKmann, 2023-09-25:
So did you resolve your Problem now? @PedroMarcos04

PedroMarcos04, 2023-09-26:
After installing the manual leveling kit, I was able to print a test.  But it still wasn't as good as it was at the beginning.  I'm thinking about putting a glass bed in place of the PEI board and not automatically leveling the bed, just adjusting the z-offset.

FabioAmorim1501, 2023-09-26:
Still a very strange situation.... 
Well it could be a too bended bed situation... Do you have any way of installing Octoprint and get a bed level chart?

coddiewomple, 2023-11-18:
I have had this problem since getting the machine 11 months ago and have tried every firmware and the elegoo advice of loosening the z axis nut screws which i find ridicilous but nothing solves it there is a major flaw with this machine. To be clear I bought a set square and corrected the upright geometry it was a few degrees out on the right my fault but elegoo offer no advice on this, I bought a laser gauge to make sure the X axis is level relative to the bed (it wasnt as set by factory so i corrected it), i replaced worn V wheels and checked for play on all axis correcting any found. Have removed the hot end so many times I can do it blindfold just cant find why the Z offset variance is always there. If you can get a first layer down its usaully a small model in the centre of the machine but try utilising the large heatbed area and the Z offset variance causes failure usually. Theres no point in having a plus or max model because its not possible to print a large model succesfully at this time due to Z offset variance. I dont know if its a HW or firmware problem but the probes are £30 each and i aint keen to find out at that price. Come on elegoo....

HonzaMiki, 2023-11-18:
@PedroMarcos04
Hello I have a Z-axis problem with my printer too, I described it more in my issue and I may have found the core of the problem, can you check my issue and give a response to it? I made a tutorial, so we can prove that more people have the same problem, so that means it is faulty firmware.
Thank You have a great day

chad-bussy, 2024-10-05:
I have the same issue here with my Neptune 3 pro. There are also tons of other bugs that I see often on this printer (eg. light turning on and off at times, randomly restoring factory settings, etc) I'd like to think that this leveling issue may have to do with saving leveling settings, judging by the previous issues I have seen on my printer regarding saving things, firmware, and such.

chad-bussy, 2024-10-05:
@HonzaMiki 
where is your tutorial?

HonzaMiki, 2024-10-06:
@chad-bussy 
Hello, actually for me problem with leveling was solved by changing the beginning of the G-code in prusaslicer, here you can discover more **https://github.com/NARUTOfzr/Neptune_3_Pro_Plus_Max/issues/78**