NARUTOfzr/Neptune_3_Pro_Plus_Max #76: ELEGOO NEPTUNE 3 PRO FIRMWARE PROBLEMS AND Z OFFSET MANUALY PROBLEMS

CAMEN3P, 2023-09-19:
A few months ago I bought an ELEGOO NEPTUNE 3 PRO  Everything was working fine until I did the update from the official ELEGOO website, (below is the link where I downloaded the latest firmware).

https://github.com/NARUTOfzr/Neptune_3_Pro_Plus_Max

1_ First the printer started with a high pitched sound in the bottom fan. It doesn't appear all the time, only when I turn it on and sometimes. But it should not appear this sound, firstly because the printer is a few months old and secondly because it did not appear before the firmware update.

2_ The MICRO SD memory that this printer came with was constantly producing errors, so I decided to buy, before the firmware update, a new micro sd memory (VERVATIN 32GB).
Also, everything worked fine until I changed the firmware. Almost in the middle of printing, the printer shows a card error. Then I said to myself:
Possibly I have to format the card. Well, so I did, I formatted it, put the files back on the card and the error message reappeared in the middle of the print. But this time, with the message (without being able to remove it) it finished printing successfully, but I couldn't access the options menu on the 3d printer screen.

With all this I first decided to contact the company that sold me the printer. Because I wanted to know if they could give me the ORIGINAL FIRMWARE THAT CAME WITH THE PRINTER. So they replied: THEY DO NOT HAVE THE ORIGINAL FIRMWARE FOR THE MACHINE AND IF I WANTED TO MAKE THE ROLLBACK I WOULD HAVE TO CONTACT ELEGOO WEB SITE.

I have seen that other people on youtube have had the same problem. 

Apparently my conclusion is that: YOU STILL DON'T HAVE A STABLE FIRMWARE, THAT'S WHY I WANT THE ORIGINAL FIRMWARE FOR THE ELEGOO NEPTUNE 3 PRO. PLEASE.

I would really appreciate if someone could help me with this problem.

_The other big problem I have is that I can't manually change the Z offset.
Why I need this: I recently did an upgrade with it: X Linear Rails and I noticed that when I do the automatic levelling, the printer nozzle pushes the bed about 2 mm downwards and when it reaches the leveling point number 6 the printer gives me an error.

I was told that this is possibly also due to the firmware.

In short, the latest firmware for the ELEGOO NEPTUNE 3 PRO is a disaster, and for those of us who can't or don't know how to modify the firmware, we can only hope that the ELEGOO technicians will do something soon. 

I hope someone can help me and that the ELEGOO software engineers or technicians can see this and do something about it soon.

gvlatko, 2023-09-20:
you are not alone. i also have issue with the Z offset constantly changing between prints.
elegoo support did nothing but send me tutorials about setting up the printer.

also the latest first randomly restarts the machine.

go do autolevel, adjust your z offset, press back and immediately go the Prepare and Z+.
the printer stops, the screen flashes and it restarts.l

waste of money and more importantly waste of my time.

HonzaMiki, 2023-10-26:
Hello, did they send you original firmware?

Relvl, 2023-11-11:
I have an stock working printer, and if you know how I can make a firmware backup and send it to you, I will be happy to help.

coddiewomple, 2023-11-18:
Having similiar problems. Right now on this firmware when i press "print" there is a short pause followed by the printer powering down twice then the screen reboots and about every other time the eeprom seems to be cleared so have to setup levelling mesh and Z offset again. This printer its my first and likely last elegoo buy. Before this restart error i was trying to solve a z offset variance issue but i spend more time bed levelling and measuring geometry than printing whats the point i only want to print a junction box for my kitvhen light and started trying to do that 3 days ago!!!. So I press the levelling button to check the Z offset and set it to 0.1mm gap lets say thats an offset of -2.00. If i exit the levelling screen so saving the -2.00 into eeprom then go back in to levelling usually the 0.1mm i set using feeling gauge has changed and you can see this on prints that usually fail due to this Z offset variance error. Then try to level the bed with that!! good luck. For the record there are no apparent mechanical problems with the machine such as probe coming loose or bad geometry i have checked and rechecked its all i seem to do this is not fun ELEPOO

HonzaMiki, 2023-11-18:
@CAMEN3P @gvlatko 
Hello I have a Z-axis problem with my printer too, I described it more in my issue and I may have found the core of the problem, can you check my issue and give a response to it? I made a tutorial, so we can prove that more people have the same problem, so that means it is faulty firmware.
Thank You have a great day

viruz82, 2024-01-23:
I have exactly the same problems with all the latest firmwares but elegoo does not seem to be interested. What can we do? It's very frustrating if you have 13h prints and the printer suddenly reboots at 98%. 👎