NARUTOfzr/Neptune_3_Pro_Plus_Max #77: Display thumbprints and elapsed time when printing from USB like when printing from SD card

gbarbehenn, 2023-10-20:
A thumbprint can be embedded in a print by: ;gimage:0`000<P000380000?<?M1C02001dH...
This thumbprint not only a representation of a print job, but other information such as layer height, and total z height.  If the job is printed from SD card, all of this is displayed.  But it isn't displayed if printed from USB, there should be an option to display the thumbprint.
Also, the elapsed print time is not displayed, there should be the option to display elapsed print time.

hindsa-nabeel, 2023-10-20:
Use this "https://github.com/Molodos/ElegooNeptuneThumbnails/releases" Cura add-on v3.4.0, all information can be set easily through menu.

gbarbehenn, 2023-10-29:
No, adding the thumbprint information to the GCODE file only works if printing from SD card.  All comment information is sent by Cura (5.4) but discarded by the Elegoo firmware when sent over the USB.
Not only the thumbprint information is discarded but also the time-to-print and maximum Z are discarded.