NARUTOfzr/Neptune_3_Pro_Plus_Max #78: Firmware Z-axis problem

HonzaMiki, 2023-10-26:
Hi
My printer:                                   Elegoo Neptune 3 Plus
Current firmware: mainboard:     1.2.5.1b
                             UI:                   1.4.2

Problem:
I bought my printer in April and last week I updated my printer to a newer firmware version, but Z-axis leveling is now minding its own-I level bed as usual but when I want to print, the nozzle start scraping bed because it is too low. I have perfectly leveled bed (you can see in photo)
[image]
All I want is to go back to my original firmware that was installed on it when I unpacked printer, but Elegoo support is stuborn and can't or don't want to send me original version they only send me link to download newer version from here https://github.com/NARUTOfzr/Neptune_3_Pro_Plus_Max
But I already try all 1.2.5.X versions on github and nothing works, all time same problem.
Can somebody help me please to get original firmware?
[image]

radimburger, 2023-11-18:
I have a same problem with Neptune 3 Max.

coddiewomple, 2023-11-18:
Same problem i just upgraded from 1.2.5.1a to see if it would fix the issue but now i have additional problems like not been able to print at all because when i press the print button the file list does not open the machine resets and also erases the eprom so need to level and set z offset again and again.... Broken, not fit for purpose. I have checked the SD card and can find no errors with it all i did was make a folder on it to move model files into to tidy things up a bit and now my printer will not print.

coddiewomple, 2023-11-18:
my machine works again after a sd card format with default allocation size selected in fact it opening the print file list seems a lot snappier however the offset issue remains for e.g. i just had to reduct it to -2.25 to -2.15 this way i can slide a 0.10mm feeler gauge in and its ok but 0.15mm gauge will not fit this seems right for petg i am using however the z offset variance causes quality issue or outright print failure.

coddiewomple, 2023-11-18:
> Hi My printer: Elegoo Neptune 3 Plus Current firmware: mainboard: 1.2.5.1b UI: 1.4.2
> 
> Problem: I bought my printer in April and last week I updated my printer to a newer firmware version, but Z-axis leveling is now minding its own-I level bed as usual but when I want to print, the nozzle start scraping bed because it is too low. I have perfectly leveled bed (you can see in photo) [image] All I want is to go back to my original firmware that was installed on it when I unpacked printer, but Elegoo support can't or don't want to send me original version they only send me link to download newer version from here https://github.com/NARUTOfzr/Neptune_3_Pro_Plus_Max But I already try all 1.2.5.X versions on github and again it scrapes bed, all time same problem. Can somebody help me please to get original firmware? [image]

May i ask did support suggest loosening the lead screw brass nut screws yet? I have some play in the lead screw brass nut its the only part i havent replaced but they do wear so its my next buy if i can face spending more money fixing this broken machine

coddiewomple, 2023-11-18:
@HonzaMiki 

May i ask did support suggest loosening the lead screw brass nut screws yet? I have some play in the lead screw brass nut its the only part i havent replaced but they do wear so its my next buy if i can face spending more money fixing this broken machine

HonzaMiki, 2023-11-18:
@coddiewomple 
Hello,
I think the problem is not mechanical, but in firmware, all problems started when I updated to a newer version (1.2.5.1b). Yesterday I did some research with my printer. I took aluminum foil and by stepping into leveling mode forced the machine to stop like 10 cm above the bed- you can't do this with paper because the sensor only detects metal. after that I checked if Z-axis is working properly by clicking arrows to make the printer go up or down - I think  I found the problem - when you are changing the Z-offset sometimes you click on arrows but the number stays the same but Z-axis physically moves. For example- I was on -0.05 and I wanted to go to 0.05, but when I got to 0 and clicked the arrow again number stayed on 0, but the Z-axis went down – so there are two 0??? And after that I tried moving the Z-axis by 1mm- I started on -0.55 after adding 1 mm it was suddenly 0.44 :D, after a while it was 5.45, so that 0.01 mm go somewhere missing and came back after a while. My friend who I suggested to buy Elegoo printer has exact same problem  (he updated to a newer version when he got the printer). I'm sending a video for better understanding. Elegoo support told me it is a faulty motherboard, I don't believe it.
Answer to your question: yes but I don't want to do it because it's not a solution to the problem and I don't want to make another problem.
Can I ask you a favor? Please tell me:
How long have you got a 3D printer?
Did your 3D printer print excellent before updating firmware?
Can you do process that I describe-If you do it, we can prove that more people have the same problem, so that means it is faulty firmware.
Thank you for answer, have a great day.

PedroMarcos04, 2023-11-18:
I gave up on fixing elegoo, I'm going to buy a raspberry pi and turn it into a klipper or octoprint.

HonzaMiki, 2023-11-18:
@PedroMarcos04 
Okay, at least thank you for your answer, have a great day.

coddiewomple, 2023-11-19:
@HonzaMiki 

I dont like the firmware as is they all have issues and sometimes i observe strange problems with number readouts not updating as you describe but the N3 plus machine i have is driving me mad for more reasons than this. I see no need to mess with foil to test sensors I truley think that is flawed test as we have a stable PEI sheet ontop of heatbed that works just fine but there does seem to be a temprature related accuracy problem with the proximity sensor explained below however this is common to all machines that use this type of sensor not just elegoo which does leave elegoo specific HW configuration or/and firmware issue including is the sensor out of tolerance or failing?

 I have a Z offset variance ranging around 0.3mm due to issues and 0.043mm of it can be explained by the sensor at a minimum which is preventing succesfull print other than luck.
 For e.g. i switch machine on today from cold and find z offset i set yesterday to achieve print success of -2.37 (incidently the highest i ever set it to date) to be to low as actual nozzle to bed gap is less than 0.05mm as cant get 0.05mm gauge under which is concerning its about to start digging in! So adjust the gap (again!!) to where it sits right (0.10mm) giving new offset value of -2.00 (not been this low for a good while). I then go to create mesh manually and find on position 1 the gap has changed its now way to large in excess of .20mm (suprise suprise) so i press home button to bring nozzle back to centre to double check my baseline and now find the same larger gap here even though seconds before i set it to be 0.10mm with a gauge. I also note the temprature of the bed is 80C since i want to work with PETG and this appears to heat the BAOLSEN Proximity Switch to around 72C measured with laser thermal device and this puts it beyond optimal operating range as per its data sheet! (repeatability of sensor is 0.04mm @ 25C). as the temprature moves away from 25C the repeatability of the sensor is affected negatively by an additional 8% on top of the 0.04mm already present due to the nature of the device, a NPN top sensing always open inductive proximity sensor.

 The only play i can find now is in the brass lead screw nuts i suspect they are worn so have ordered replacements i had to guess about as Elegoo even within the statutory warranty period refuse to supply these parts stating "not available". Until i replace these and grease them i cannot test further to check the sensor accuracy. The cost of a replacement proximity sensor starts at £20, cost of brass lead screw nuts another £20 inclduing grease. I cant continue this path of fixing broken elegoo printer. 

@PedroMarcos04 may have the solution unless better more robust firmware support is forthcoming and i doubt it will be given the series 4 has been released now and they moved to Klipper.

florianolg, 2023-12-08:
Neptune 4 Plus  - FW version - 1.2.2.2

What I'm seeing may be similar to what you are talking about. 
Z tends to drift - I've been printing the same part for a few days now - about 50 copies. (Lucky the part tolerance does not matter to much for Z) 
Part is only 5mm thick (- will vary regularly .5mm or up to full 1mm - when that happens nozzle is rubbing on plastic - its bad)

When the failure is small you can see layers that plastic squeezed out - after those print if I don't re-zero Z, the next prints will be even worse.

When I go to Level Z (with Paper) I can see the nozzle is too close to the bed - If I start and stop a print - Re-check Z (now it's too far) - So Re-Zero again.

I get a few perfect prints then again; the Z will drift.

On one occasion when moving the Z to set the offset - with the pad - the numbers would not change on the pad - but it was physically moving.

Not sure if this helps anyoone .. but for me the problem appears to be Z drift.

coddiewomple, 2023-12-24:
I emailed elegoo ages ago about this and after sending Elegoo evidence of the problem they asked for and photos i took of the proximity sensor which seems to show up heat markings but not the naked eye i.e. only visible in the photo taken. 

Elegoo sent me a replacement proximity sensor and connector board for it that was updated, they have removed two 100 ohm resistors from the proximity sensor feed lines on the board but i dont know why they just sent it so i fitted them both togethor at same time. The replacement proximity sensor was almost identical to the old one, same make and model markings just a few extra lines pertruding on its moulding. Seems to have tightend the variance up a bit but unable to test as having FW issues now and cant seem to make use of the big bed printer i dont see the point of it really in this condition. If i print with M420 active or inactive it seems to make very little difference i begin to suspect this machine is all gimmick a press and print PLA only affair because if you try anything else it fails bad.

chriskeeble, 2023-12-24:
We have two Elegoo Neptune 3 Plus units and both were recently updated to latest firmware.

This affected the z-axis, but going through all steps of the bed levelling procedure (physical, manual, automatic) and adjusting the z-offset configured everything perfectly.

coddiewomple, 2023-12-24:
@chriskeeble 

The latest version is the "C" version that comes with UI 1.4.3 all linked below. Is that the version your machines are running without issue?

https://github.com/NARUTOfzr/Neptune_3_Pro_Plus_Max/releases/tag/firmware_1.x.5.1c-UI_V1.4.3-20230707

coddiewomple, 2023-12-24:
[image]
[image]

coddiewomple, 2023-12-24:
For the record here is info on a similiar proximity device made by Panasonic a GX-H12AI

https://na.industrial.panasonic.com/products/industrial-automation/industrial-sensors/series/71523/model/75566

[image]

alexose, 2024-01-14:
Posting here because I'm running into a similar issue.  My Neptune 3 Plus seems to have forgotten it's zero position for the Z axis.  It pretty much destroyed by PEI sheet before I noticed what was going on.

I didn't change anything with the machine, although I did swap out the SD card.  Very odd.

viruz82, 2024-01-23:
I have exactly teh same issue with my Neptune 3 Pro.

viruz82, 2024-02-10:
@NARUTOfzr is not interested to fix this problem?

HonzaMiki, 2024-02-11:
For me is a problem somehow fixed and I would never expect this easy solution. In Prusaslicer I changed the beginning of the G-code, so before the printer sets its zero points it will first heat the bed to the final temperature, after that, it will set its zero points. Now it’s working well again.

Here is the beginning of the G-code if somebody wants to try it:

M413 S0 ; zakázat obnovu po výpadku proudu
G90 ; použít absolutní souřadnice
M83 ; režim relativního extruderu
M140 S[first_layer_bed_temperature] ; nastavit konečnou teplotu postele
M104 S120 ; nastavit dočasnou teplotu trysky k zabránění vytekaní při homingu a automatickém nivelizaci postele
G4 S10 ; umožnit částečné zahřátí trysky
M190 S[first_layer_bed_temperature] ; počkat, až se teplota postele ustálí
G28 ; domovská pozice všech os až po dosažení teploty postele
;G29 ; spustit síťovou strukturu
M420 S1 ; načíst síťovou strukturu
G1 Z50 F240
G1 X2 Y10 F3000
M104 S[first_layer_temperature] ; nastavit konečnou teplotu trysky
M109 S[first_layer_temperature] ; počkat, až se teplota trysky ustálí
G1 Z0.28 F240
G92 E0
G1 Y140 E10 F1500 ; první přítlak trysky
G1 X2.3 F5000
G92 E0
G1 Y10 E10 F1200 ; druhý přítlak trysky
G92 E0

viruz82, 2024-02-26:
@HonzaMiki wow, great! I can confirm that no everything works as it should. Many thanks!

NiklasTheis, 2024-02-26:
@HonzaMiki or @viruz82 or anybody else. Do you know if this could also work with UltiMaker Cura (5.3.1 or newer)?

HonzaMiki, 2024-02-26:
Hello @NiklasTheis , I don't have any experience with any type of Cura, but in Elegoo Cura I think it is preheating the bed first as I was describing, but I'm not 100% sure about it. You can try to search on the Internet.
I installed Elegoo Cura and here is the beginning of the G-code:

G28 ;home
G92 E0 ;Reset Extruder
G1 Z4.0 F3000 ;Move Z Axis up
G92 E0 ;Reset Extruder
G1 X1.1 Y20 Z0.28 F5000.0 ;Move to start position
G1 X1.1 Y80.0 Z0.28 F1500.0 E10 ;Draw the first line
G1 X1.4 Y80.0 Z0.28 F5000.0 ;Move to side a little
G1 X1.4 Y20 Z0.28 F1500.0 E20 ;Draw the second line
G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up

VitorMigS, 2024-12-26:
> **(HERE IS SOLUTION THAT WORKED FOR ME- change the beginning of the G-code, so before the printer sets its zero points it will first heat the bed to the final temperature, after that, it will set its zero points. Here is beginning of G-code for **prusaslicer**:
> 
> M413 S0 ; disable recovery after power failure G90 ; use absolute coordinates M83 ; relative extruder mode M140 S[first_layer_bed_temperature] ; set the final temperature of the bed M104 S120 ; set a temporary nozzle temperature to prevent bleed-off during homing and automatic bed leveling G4 S10 ; allow partial heating of the nozzle M190 S[first_layer_bed_temperature] ; wait for the temperature of the bed to stabilize G28 ; home position of all axes until bed temperature is reached ;G29 ; start the network structure M420 S1 ; load the network structure G1 Z50 F240 G1 X2 Y10 F3000 M104 S[first_layer_temperature] ; set the final temperature of the nozzle M109 S[first_layer_temperature] ; wait for the nozzle temperature to stabilize G1 Z0.28 F240 G92 E0 G1 Y140 E10 F1500 ; first nozzle pressure G1 X2.3 F5000 G92 E0 G1 Y10 E10 F1200 ; second nozzle pressure G92 E0**
> 
> Hi My printer: Elegoo Neptune 3 Plus Current firmware: mainboard: 1.2.5.1b UI: 1.4.2
> 
> Problem: I bought my printer in April and last week I updated my printer to a newer firmware version, but Z-axis leveling is now minding its own-I level bed as usual but when I want to print, the nozzle start scraping bed because it is too low. I have perfectly leveled bed (you can see in photo) ![Obrázek WhatsApp, 2023-10-24 v 22 08 35_d21672a9](https://private-user-images.githubusercontent.com/149058128/278282280-d41e16eb-4703-465d-9fd0-df34e1b09c58.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzUyNDgxMTgsIm5iZiI6MTczNTI0NzgxOCwicGF0aCI6Ii8xNDkwNTgxMjgvMjc4MjgyMjgwLWQ0MWUxNmViLTQ3MDMtNDY1ZC05ZmQwLWRmMzRlMWIwOWM1OC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIyNlQyMTE2NThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zYjMwN2M3ZTgzNWIwYmYwNzZjYmY5ZDQxODY3YjA4MTgxZTg1OGQwMmY1NjQ1N2NhMTAxNzgyMTE2NjQ3ZjM5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.mUzRomWRnyPgdTnki8QrceLENf9qziBFMfOVMC4Hy1E) All I want is to go back to my original firmware that was installed on it when I unpacked printer, but Elegoo support can't or don't want to send me original version they only send me link to download newer version from here https://github.com/NARUTOfzr/Neptune_3_Pro_Plus_Max But I already try all 1.2.5.X versions on github and again it scrapes bed, all time same problem. Can somebody help me please to get original firmware? ![Obrázek WhatsApp, 2023-10-17 v 16 34 45_715ecff4](https://private-user-images.githubusercontent.com/149058128/278285451-20b33caf-adeb-456e-9b7f-0594eb05f4d2.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzUyNDgxMTgsIm5iZiI6MTczNTI0NzgxOCwicGF0aCI6Ii8xNDkwNTgxMjgvMjc4Mjg1NDUxLTIwYjMzY2FmLWFkZWItNDU2ZS05YjdmLTA1OTRlYjA1ZjRkMi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIyNlQyMTE2NThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03NjE3YTU2OGVjNWY5NGE3MDNkMmI4NWRiNmZkMDk3YWNmY2E1NjVhZDQyN2ZiOTkzM2ViNDgxNjBlYjRhNDMwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.K9yDK2JsRPbS1IhGnPiOAF2J625mBXY0pywUzyfOz3Y)
> 
> **!!!UPDATE!!!** More info to my problem: I think the problem is not mechanical, but in firmware, all problems started when I updated to a newer version (1.2.5.1b). Yesterday I did some research with my printer. I took aluminum foil and by stepping into leveling mode forced the machine to stop like 10 cm above the bed- you can't do this with paper because the sensor only detects metal. after that I checked if Z-axis is working properly by clicking arrows to make the printer go up or down - I think I found the problem - when you are changing the Z-offset sometimes you click on arrows but the number stays the same but Z-axis physically moves. For example- I was on -0.05 and I wanted to go to 0.05, but when I got to 0 and clicked the arrow again number stayed on 0, but the Z-axis went down – so there are two 0??? And after that I tried moving the Z-axis by 1mm- I started on -0.55 after adding 1 mm it was suddenly 0.44 :D, after a while it was 5.45, so that 0.01 mm go somewhere missing and came back after a while. My friend who I suggested to buy Elegoo printer has exact same problem (he updated to a newer version when he got the printer). I'm sending a video for better understanding. Elegoo support told me it is a faulty motherboard, I don't believe it.
> 
> Here is tutorial if you want to find out if you have this problem:
> 
> 1. In the prepare section move the printer head 20 cm above (or like you want),
> 2. take aluminum foil and go into leveling mode, sensor wont detect paper, metal only
> 3. now you should be able by holding aluminum foil under the sensor to force the printer head to stop 10-20 cm above the bed
> 4. choose 1mm, now when you click down/up you can find that 0.01 mm goes missing sometimes (as in the video I sent)
>    Additional steps:
[... cut ...]
> I will be really grateful if as many people as possible try this-If you do it, we can prove that more people have the same problem, so that means it is faulty firmware.
> 
>  Video.WhatsApp.2023-11-15.v.21.29.08_801d3649.mp4 
>  Video.WhatsApp.2023-11-15.v.21.29.08_f27d618e.mp4

Hello everything is fine?
I have a Neptune 3 and am experiencing the same issues with the table automatically leveling.
Do you make this change in gcode to define the zero points before printing? Or when turning on the printer does it still need self-leveling?