NARUTOfzr/Neptune_3_Pro_Plus_Max #82: Print > Files corrupts / merges filenames on SD card

chriskeeble, 2023-12-18:
When viewing files to print on the SD card, in root folder, the filenames are corrupted.

Here are the 3 files as seen on the Neptune 3 Plus:

Screenshot of Neptune Screen
[image]

Screenshot from Windows
[image]

- Firmware Version 1.2.5.1b
- UI Version V1.4.2 
(V1.4.2-1.x.5.1b - 20230520, V1.4.2-20230520)

(I love that we can now see the whole filename in this firmware, but each filename should be separate).

hindsa-nabeel, 2023-12-19:
Try keeping file names less than 30 characters.
This problem is already reported (in Chinese) here:
https://github.com/NARUTOfzr/Neptune_3_Pro_Plus_Max/issues/62

chriskeeble, 2023-12-30:
> Try keeping file names less than 30 characters. This problem is already reported (in Chinese) here: #62

Thank you - that does appear to be the same bug.  I've also experienced freezing after pressing "Print" with some files, as you described, but I believe this is may be a different issue.

Thanks again.

chriskeeble, 2023-12-30:
Duplicate of #62

closed 2023-12-30 by chriskeeble