NARUTOfzr/Neptune_3_Pro_Plus_Max #86: Neptune 3 plus toggles still not working properly

Panda2D, 2024-02-29:
I just updated my Neptune 3 plus with latest firmware hoping to have toggle switches in settings to work but they still are random or not working.

For the led I have to tap the toggle twice to get it to be on and show correctly wether it's on or off.

For power recovery the toggle just says on and off but the feature never really works. It only recovered ones in the year I had the printer.

After every print the setting just reset. They don't stay on.

hindsa-nabeel, 2024-03-05:
This sounds like a hardware issue. It may be a faulty touch screen, or a bad connection/component on the board, or leakage/dampness, or the device is not assembled properly causing pressure on the screen or the board.
Try to disassemble the casing of the hand-held device (see if it works), inspect the connections and components (fix if necessary), clean it and reassemble.
Good luck.

Der-Lorax, 2024-05-07:
It's not an Hardware Issue. It's the same on my printer and because of the recovery, sometimes the print looks horrible.

Panda2D, 2024-05-08:
Everything else works. Only the power loss recovery and built in LEDs is acting up. 

The light will turn on after two presses at first and then work normally until the printer powers down.

The power loss recovery will toggle on and off in the UI but I don't think it's doing anything.

Der-Lorax, 2024-05-08:
I'm having trouble with my printer stuttering in sharp curves. That's because my SD card isn't fast enough. Without power loss recovery enabled, the printer isn't saving data (like the extruder's position) to resume printing if the power goes out. However, with power loss recovery enabled, the printer sometimes creates imperfections (like blobs) on the outside of the object and my print looks horrible.