NARUTOfzr/Neptune_3_Pro_Plus_Max #87: ESP8266 WI-FI module support for Neptune_3_Pro_Plus_Max

hindsa-nabeel, 2024-03-05:
It has been since more than 8 months when @NARUTOfzr promised to help with this issue, I hope he will deliver an update soon.

cochonez, 2024-03-08:
I think it's a cheap china printer issue. They probably just dont want to invest any money into "old printers".

Nanaharu, 2024-03-18:
I asked the Official Technical Support in October if they still work on a new firmware for this printer. 
They responded with "yes".
5 months passed by now and nothing happened...
@hindsa-nabeel I switched to Klipper a few days ago and I'll never return to stock firmware.
Could this be an option for you?
A cheap RPi Zero 2 W (~20€) will do the job easily.

hindsa-nabeel, 2024-03-19:
I was also contemplating Klipper; for connectivity and speed advantages, since I have RPi 4. 
But your idea is interesting, since it may be possible to place the RPi Zero inside the printer's enclosure.
It would be great if you could state the challenges/limitations you met during the process. 
Nevertheless, Elegoo needs to honor its promises to maintain customers' trust. even though they moved to Klipper in the Neptune 4 series.

hindsa-nabeel, 2024-05-01:
I was happy to see @NARUTOfzr responding to an issue on this forum today, I was concerned that something might have happened to him. But thank Goodness he is alive and well.
I hope he can dedicate some time for the inclusion of ESP8266 WI-FI module support in the firmware.