NARUTOfzr/Neptune_3_Pro_Plus_Max #89: Firmware 1.1.5.1c (Neptune 3 Pro) - Filament Sensor not working

T1gg0r, 2024-03-29:
Greetings,

I upgraded the firmware on my Neptune 3 Pro from 1.1.5.1a to 1.1.5.1c (the "latest" Release) including the appropriate screen firmware.
Flashing worked well, I recalibrated everything and started a print. I was expecting the filament to run out in this print, but I also expected the filament sensor to pause the print.
The filament did indeed run out, but the print was not paused. The filament sensor was activated in the display settings of the printer.
But when looking with an eeprom editor in octoprint, the checkbox for the filament sensor is missing (which was there on version 1.1.5.1a).

Also, as I made a backup of the eeprom values before the upgrade, here is the file difference:
1.1.5.1a: "filament_runout": {"command": "M412", "params": {"D": null, "H": null, "S":true}}
1.1.5.1c: "filament_runout": {"command": "M412", "params": {"D": null, "H": null, "S": null}}

I will now downgrade to 1.1.5.1b and check again for the sensor. 
Btw, the 1.1.5.3 version is actually older than the 1.1.5.1b & c versions? Slightly confusing.

T1gg0r, 2024-03-29:
Well, small update:
I tried something before the downgrade. I edited the backup I made of the eeprom settings and changed the filament runout param "S" to "true". Then I imported it, which atleast fooled the eeprom editor. But even after a reboot of the printer and starting a print (without filament in the sensor), no stopage occured.

After the downgrade to 1.1.5.1b, the filament sensor did still not work.
Downgrading to 1.1.5.1a again...

Aaaaand it did work in Version 1.1.5.1a with display version 1.4.1. So I'm back to start.

vehystrix, 2024-10-24:
Ok so for some reason in the 1.1.5.1b release the filament sensor is disabled. [See this commit](https://github.com/NARUTOfzr/Elegoo-Neptune-marlin2.1.1/commit/9631b7f5eb60b9c94fca12384662718f3b1ef060#diff-bb298d18dac4268225a7a53a5ad24844472530bdc63f6a98a2ba8d8158b75080R1834)
So downgrading the firmware to 1.1.5.1a will indeed fix it, or you could rebuild from source with the sensor enabled again