NARUTOfzr/Neptune_3_Pro_Plus_Max #92: Please new firmware release: UI 1.5.x and Marlin 2.1.2.2

nevergone, 2024-05-16:
I would like to request an official release of UI 1.5.x and Marlin 2.1.2.2 for Neptune 3 series.

我想请求 Neptune 3 系列的 UI 1.5.x 和 Marlin 2.1.2.2 正式版本。

Thanks!

https://github.com/NARUTOfzr/Marlin-2.1.2.2

gbarbehenn, 2024-05-22:
Please add the UI source files to the Neptune 3 Pro/Plus/Max.  The necessary source files for DGUS are present in the Neptune 3 screen firmware repository, but not in the Neptune 3 Pro/Plus/Max  screen firmware repository.

ellensp, 2024-05-26:
@gbarbehenn Its not a DGUS display on those model, its a TJC screen and they do provide the source HMI file https://github.com/NARUTOfzr/Neptune_3_Pro_Plus_Max/tree/main/Source%20Code/LCD-TJC/V1.4.2

hindsa-nabeel, 2024-07-10:
Would this help:
https://github.com/Hummtaro/Elegoo-Neptune-marlin/releases