NARUTOfzr/Neptune_3_Pro_Plus_Max #93: M0 Gcode command does not work

WestInSpace, 2024-06-04:
When giving the printer the M0 geode command it doesn't work. It just ignores the command and continues printing. I know M600 works to change filament, but I want an unconditional pause for a certain amount of time rather than based on loading filament so that I can put magnets in my 3d prints.

chriskeeble, 2024-06-04:
I have successfully used the filament change code to pause, insert nuts (magnets would work just the same) and then unpause without changing filament.

It makes no difference apart from what it says on the screen.

WestInSpace, 2024-06-04:
Dropping magnets in was just an example. (A bad example I guess) what I
actually want to do is home the toolhead then pause for a certain amount of
time then home again and start the print. I want to do this to allow the
inductive sensor to heat up so I can get more consistent prints and
z-offset. Heat affects inductive sensors a lot, I have measured as much as
0.13mm from the bed difference between a cold and hot inductive sensor. So
I want it to heat up before homing before a print. If that makes sense.