NARUTOfzr/Neptune_3_Pro_Plus_Max #96: Original MicroSD Image

Fasteddiechu, 2024-11-09:
I was trying to clean up printed gcode files and left the original folders alone. I must have inadvertently deleted something else because I can no longer print using the MicroSD card. The printer does not display the files to be printed. Is it possible to download the original card image so that I can see what files are missing?

Daomephsta, 2024-11-10:
Your problem isn't missing files or folder structure, as there are no such requirements. It's likely an issue with the names of the GCode files. If a filename is too long or contains non-ASCII characters, it's ignored, or in some situations all the files are ignored.

I can't be more precise than that, as I haven't managed to find precise information about what filename lengths and characters Marlin (basis of Neptune 3 firmware) supports.
Doing some testing with mine to see if I can figure out filename length limit.

Daomephsta, 2024-11-10:
Experimenting, I'm not sure there is a fixed limit?
I run into issues (specific files not displaying, all files not displaying) not when I exceed a specific length, but in general with long filenames and a lot of files.

Fasteddiechu, 2024-11-11:
Thank you-thank you-thank you!

The names were too long. I renamed it shorter and it shows up!

Appreciate your help!

Eddie

Fasteddiechu, 2024-11-11:
Thanks again. Saves me running my PC all the time to print long projects
through the USB.

Eddie