NARUTOfzr/Neptune_3_Pro_Plus_Max #98: printing

GhoostAlien, 2025-02-21:
so i have been printing since dec 2024.   n they been good prints.    start on monday i have been getting prints that not printing
correctly   the temp is 220 and 55.  i clean the plate every time i use my printer so its always clean.    the prints is very soft, (where the it should be hard as its printing)    i have not changed anything as temp or anything,   it print like its missing spots.  there is no holes to make is missing, should be solid pieces...  also it looks like its printing very thin...   not like how it used to..  also the there is motor sounds like maybe grinning.   i can send u a audio clip or on phone when it comes up..     this looks like its in the motor, like a clog..  if it is how can i fix this.. 

i have a picture of how its coming out.   i would like to talk to someone my voice please...  email is always hard to describe things!

chriskeeble, 2025-02-21:
You should probably contact Elegoo support - I've heard that they are very good with product issues / faults.

https://www.elegoo.com/en-gb/pages/contact-us

GhoostAlien, 2025-02-21:
i have contqct them  and them gave me to links..  one was a video...   https://www.youtube.com/watch?v=pii84q9hqLI   the other was to post someone on here..      this doesnt make any sense!!!1

KoBach276, 2025-02-21:
Post your pictures on the elegoo subreddit and ask for help there. Community is pretty helpful there.

KoBach276, 2025-02-21:
https://www.reddit.com/r/elegoo/

GhoostAlien, 2025-02-21:
thank you   posted on there..