NARUTOfzr/Neptune_3_Pro_Plus_Max #101: SOLUTION: All Firmware Update Issues Caused by EEPROM Structure Changes - Factory Reset Required

EsurmaGm, 2025-07-13:
# SOLUTION: All Firmware Update Issues Caused by EEPROM Structure Changes - Factory Reset Required

## Problem Summary
Multiple users are experiencing critical issues after firmware updates including:
- Z-endstop sensors not working/crashing into bed
- Bed leveling failures
- Settings randomly resetting
- "EEPROM datasize error" messages
- Filament sensors not functioning
- Print quality issues

**Root Cause Discovered:** Elegoo changed the EEPROM memory structure between firmware versions, but old EEPROM data is being misinterpreted by new firmware, causing system instability and feature failures.

## Technical Explanation
When firmware is updated, the EEPROM (stored settings) layout often changes:
- Memory addresses for settings get remapped
- Data structure sizes change
- New variables are added/removed
- Variable types may change

Old EEPROM data at memory address X might have been "Z-endstop sensitivity" in the old firmware, but the new firmware expects "fan speed setting" at that same address. This causes the printer to interpret stored values incorrectly.

## The Solution: Factory Reset EEPROM After ANY Firmware Update

### Method 1: Via Printer Menu (Easiest)
1. Navigate to **Settings → Factory Reset** (or similar menu option)
2. Confirm the reset
3. Printer will restart with clean EEPROM using firmware defaults
4. Re-calibrate your Z-offset and bed leveling

### Method 2: Via Terminal/OctoPrint 
```gcode
M502    ; Load firmware defaults to RAM
M500    ; Save defaults to EEPROM
```

### Method 3: Via G-code File
Create a text file named `reset_eeprom.gcode` with the following content:
```gcode
M502    ; Load firmware defaults
M500    ; Save to EEPROM
M117 EEPROM Reset Complete    ; Display message
```
Print this file from SD card.

## Issues This Solves

This solution directly addresses:
- #59 - "EEPROM datasize error" 
- #71 - Printer resets and settings lost
- #80 - "loss of EEprom data FW 1.2.5.1b"
- #89 - Filament sensor not working after update
- #76 - Major firmware problems and Z-offset issues
- #78 - Z-axis leveling problems after firmware update

## Why This Wasn't Obvious

1. **Fresh printers** (minimal EEPROM customizations) appeared to work fine with new firmware
2. **Heavily customized printers** (lots of stored settings) experienced the worst issues
3. **Elegoo's firmware update instructions** failed to mention this critical step
4. **Users assumed** firmware updates would handle EEPROM compatibility automatically

## Prevention: Proper Firmware Update Procedure

**Before updating firmware:**
1. Note down your current settings: `M503` (or write down Z-offset, bed leveling values, etc.)
2. Backup EEPROM if possible

**After updating firmware:**
1. **Always perform factory reset** (M502 + M500)
2. Re-enter your custom settings
3. Re-calibrate bed leveling and Z-offset
4. Test all printer functions

## Technical Evidence

Users have documented EEPROM structure changes between versions:
- v1.1.5.1a: `"filament_runout": {"command": "M412", "params": {"S":true}}`
- v1.1.5.1c: `"filament_runout": {"command": "M412", "params": {"S":null}}`

This confirms the EEPROM layout changes between firmware versions.

## Message for Elegoo

**Please update your firmware installation instructions** to include:
> ⚠️ **CRITICAL STEP:** After flashing new firmware, perform a factory reset via Settings → Factory Reset or send M502 + M500 commands. This prevents compatibility issues with stored settings from previous firmware versions.

## For Future Users

If you experience **any** strange behavior after a firmware update:
1. Try factory reset first (M502 + M500)
2. Recalibrate your printer
3. Most "firmware bugs" are actually EEPROM compatibility issues

---

**This issue affects dozens of users across multiple GitHub issues. A single factory reset solves problems that users have been struggling with for months.**

**Verified Solution:** Tested on Neptune 3 Pro with firmware update from working state → broken Z-endstop → factory reset → fully functional.

EsurmaGm, 2025-07-13:
closed as complete

closed 2025-07-13 by EsurmaGm

rolandjutai, 2025-08-14:
Would be good to confirm this solves all the issues, because it hasn't... the below issue still persists for us. We do EEPROM resets. Wonder if the bed mesh points also changed...

https://github.com/vehystrix/Elegoo-Neptune-marlin/issues/3